supermerill/SuperSlicer #628: Stuck compiling on fedora 33

Tuxomx, 2020-11-03:
### Operating system type + version
fedora 33

Trying to compile on fedora 33 even when I have wxWidgets installed i get the error about not finding that library. 

Then tried to use the deps-linux.cmake file to build and install locally the "missing" dependencies, that is, inside SuperSlicer/deps

mkdir build
cd build
cmake ..
(after trying next commands I issued a make command with no error messages)

and wait, after this i just issued 
cmake .. -DSLIC3R_STATIC=1-DCMAKE_PREFIX_PATH=/usr/local/src/SuperSlicer/deps/build/usr/local
but the same error
and then 

cmake .. -DSLIC3R_STATIC=1-DCMAKE_PREFIX_PATH=/usr/local/src/SuperSlicer/deps/build/dep_wxwidgets-prefix/src/dep_wxwidgets/usr/local

in all cases i am getting this at the end:

CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS) (Required is at least version "3.0")
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindwxWidgets.cmake:1008 (find_package_handle_standard_args)
  src/CMakeLists.txt:43 (find_package)

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/SuperSlicer/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/SuperSlicer/build/CMakeFiles/CMakeError.log".

Checking the two last files (cmakeOutput.log and CMakeError.log says nothing about wxWidgets but stuff about other things and cereal not being found.

If I run just "cmake .." I get the same error message, one of the lines say 
 -- SLIC3R_STATIC: 1-DCMAKE_PREFIX_PATH=/usr/local/src/SuperSlicer/deps/build/dep_wxwidgets-prefix/src/dep_wxwidgets/usr/local
-- SLIC3R_WX_STABLE: 1
-- SLIC3R_STATIC: 1-DCMAKE_PREFIX_PATH=/usr/local/src/SuperSlicer/deps/build/dep_wxwidgets-prefix/src/dep_wxwidgets/usr/local
-- CMAKE_PREFIX_PATH: ../deps/ (from cache or command line)

so looks like I need to do something like "make clean" but I think for this I just have to erase the build subdirectory and start over again, and of course looks like I will get the same results then.

SALUDOS
ALVARO

supermerill, 2020-11-03:
1)  be sure you have correctly generated the deps. Forget about your installed wxwidget.
2) use something like -DCMAKE_PREFIX_PATH=/usr/local/src/SuperSlicer/deps/build/destdir/usr/local
3) do not forget to put a space between '1' and '-DCMAKE'

jin-eld, 2021-06-01:
> be sure you have correctly generated the deps. Forget about your installed wxwidget.

Could you please elaborate a bit more on that? Why wouldn't superslicer use the installed wxwidget libraries?

Trying to compile SuperSlicer on Fedora 33 as well, installed all wxBase/wxGTK devel packages, I do have `wx-config` and `wx-config-3.1` in `$PATH`, still I get:

```
CMake Error at src/CMakeLists.txt:50 (message):
  

  Could not find wxWidgets 3.1.

  Hint: On Linux you can set -DSLIC3R_WX_STABLE=1 to use wxWidgets 3.0
```

supermerill, 2021-06-01:
The wxwidget libs should be on superslicer/deps/build/destdir/usr/local/....
and you need to rename one of them

jin-eld, 2021-06-01:
> The wxwidget libs should be on superslicer/deps/build/destdir/usr/local/....
> and you need to rename one of them

Well, that's the first question.... why? :) Are those patched wx versions or why can't the system versions be used?

The second question - its not quite clear to me what you mean, so I should go to `deps`, then `mkdir -p build/destdir/usr/local/`, then `cp -a wxWidgets build/destdir/usr/local/` and then `cd build && cmake ../..` and cmake will find it?

supermerill, 2021-06-02:
I'll create a build script.

BosenSwe, 2021-08-31:
I have the exact problem with the build in Fedora 34.

supermerill, 2021-08-31:
try `BuildLinux.sh -dsi`

BosenSwe, 2021-08-31:
I just get:
`dpkg-query: no packages found matching libgtk*
FOUND_GTK2=)
Error, you must install the dependencies before.
Use option -u with sudo
`

supermerill, 2021-08-31:
oh, right, sorry for the false solution.