supermerill/SuperSlicer #631: Wrong chronology preview

Yordan1976, 2020-11-03:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
2.2.54

### Operating system type + version
_What OS are you using (and version)_
Win 10 Pro

When I use sequential printing / Complete individual objects the chronology preview doesn't reflect this. 
Maybe it is related to this issue, that the remaining time to completition in OctoPrint is also wrong until it reaches the last object.

[image]

supermerill, 2020-11-03:
It's revamped in the next version with the new prusa gcode-viewer, I'll check if it stay that way.

supermerill, 2020-11-04:
It's fixed in the 2.3.55 version

closed 2020-11-25 by supermerill