supermerill/SuperSlicer #632: units wrong for RepRap firmware

Phil1988, 2020-11-03:
### Version
2.2.53

### Operating system type + version
Win10 x64

First of all: 
Thanks for this great software.
This slicer is really nice and I never had issues on Marlin.

Now I am using this the first time on a RepRap Firmware Printer had a really hard time on tuning a 3D printer because the SuperSlicer has some messed up units. (in Machine limits settings)

These settings:
[image]

generates this gcode:
`M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 T1500 ; sets acceleration (P, T), mm/sec^2
M566 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M107
G10 P0 R200 S195 ; sets the standby temperature
M201 X1000.00 Y1000.00 Z1000.00 E5000.00       ; set accelerations (mm/s^2)`

The result were stuttering prints.

The mistake is on the jerk setting, as M566 is mm/min not per second in RepRapFirmware:
[image]

Source:
https://duet3d.dozuki.com/Wiki/Gcode#Section_M204_Set_printing_and_travel_accelerations

So this has to be calculated differently or the unit in the fields should be renamed to mm/min instead of mm/s.

M203 is also mistakenly commented in the GCODE.
This resulted in more confusing of myself:
M203 X500 Y500 Z12 E120 ; **sets maximum feedrates, mm/sec**

It is in **mm/min**

I hope you can take a look and fix this issue.

Thanks and all the best :)

supermerill, 2020-11-03:
kind of duplicate of #440 
the unit has been fixed in the gui since 2.3.53.2
the rest is fixed in 2.3.55 (not released yet)

Phil1988, 2020-11-03:
Thanks for the response.

The latest released version for windows I saw ist 2.**2**.53.
Is there really a 2.3.53.2 or is it a typo?

The SuperSlicer tells me its version is 2.2.53, even tho the extracted folder is named SuperSlicer_2.2.53**.3**_win64_20201005.

supermerill, 2020-11-03:
I updated the displayed name from the 2.2.54.1 version
first two number are the same as the prusa release, it's to know from what prusa version it's made.
second number is the release by me, and the third is the path number, for bugfixing.

current stable release: 2.2.53.3
current beta release: 2.2.54.2

And I am going to publish the merge with the prusa 2.3 alpha later this week.

closed 2020-11-25 by supermerill