supermerill/SuperSlicer #634: Complete individual Objects -> too tall

Yordan1976, 2020-11-04:
I don't know, if it is a bug or if I am just expecting another behavior.

### Version
2.2.54
### Operating system type + version
Win10 Pro

### 3D printer brand / version + firmware version (if known)
Caribou MK3s 320 (similar to Prusa MK3s)

### Behavior
The extruder clearance settings are radius=45 and height=20

My expectation is, that objects should be printable individually if they max 20mm high (not the case here) OR outside the radius, especially on the y-axis (which is the case here). If the object on the back is completed first, the will be no crash when printing the object on the front. Or am I missing something?

[image]

supermerill, 2020-11-04:
The slicer doesn't know how your printer is built.
If you're sure there is no problem, then put a big value in z clearance for this print.

Yordan1976, 2020-11-04:
Ok, how will it move to the next object? x/y first, z when in place?

supermerill, 2020-11-04:
yes

Yordan1976, 2020-11-04:
Ok, thanks!

closed 2021-04-27 by supermerill