supermerill/SuperSlicer #636: the hotend doesn't heat up with Super slicer despite on  it is in the G-Code

mafotaso, 2020-11-05:
### Version
V2.2.53

### Operating system type + version
Win10 + v1909

### 3D printer brand / version + firmware version (if known)
Custom Printer + SKR V1.3 + Marlin 2.2.7.1
### Behavior
* Not heating in the hotend.
- I just have install SuperSlicer and copy paste the profiles from PrusaSlicer to SuperSlicer folder. 
-I made a piece, without problems yesterday. 
- Today in order to adjust "compensation XY" I have tested print squares. First try, the extruder stopped in the second loop of the Skirt but the printer was going on with the piece. Then I tought "okey.. it is maybe a problem with upload  of the g-code to SD" so I exported again the G-Code, and the print worked fine. 
- Then, I try again the square to adjust better "compensation XY" and the printer started to print without move the extruder from the beginning and I realized that the hotend's temperature was set in 0ºC!! So Marlin setting prevented move extruder for Tmin extruder. 
- The hotend works, perfectly, with PrusaSlicer too, and I have compared SuperSlicer&PrusaSlicer's G-Codes:
Super Slicer starts with (I have shaded the differences):
_M107
M190 S60 ; set bed temperature and wait for it to be reached
**G10 P0 R200 S215** ; sets the standby temperature
**G10 P0 S215** ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
M150 B0 R255 U0 W255    ; change colour screen
**G10 P0 S215** ; set temperature
M116 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0_

And PrusaSlicer generates:
M107
M190 S60 ; set bed temperature and wait for it to be reached
**M104 S215** ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
M150 B0 R255 U0 W255    ; change colour screen
**M109 S215** ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0

In addition I just copy thoses Prusa G-code lines and paste to the SuperSlicer and the print works perfectly then I have re-write the original SuperSlicer G-code and the print mades the same problem, it set 0ºC in hotend. This seems like it doesn´t know to read the G10 command.

* _Steps needed to reproduce the problem_
Only generate the G-Code

* _Actual Results_
Headache
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
No needed.

MarcPot, 2020-11-05:
What happens if you set the gcode flavor to Marlin instead of reprap?

supermerill, 2020-11-05:
please upgrade to 2.2.43.3 or 2.2.54.2 if you need the new features it has (or the latest nighlty).

mafotaso, 2020-11-05:
> What happens if you set the gcode flavor to Marlin instead of reprap?

in others softwares if you select Marlin, it change the speeds configurates when you generate the G-code. The people always have selected rep-rap to prevent it but I think that with Superslicer you have to select Marlin yes or yes and then change yourself the speeds and acceleration again in Super slicer whatever you have in firmware Marlin

> please upgrade to 2.2.43.3 or 2.2.54.2 if you need the new features it has (or the latest nighlty).

I downloaded V2.2.53 in order to this advice "The current stable release is 2.2.53.3 (see below), not this one." so I installed this stable version.

supermerill, 2020-11-05:
sorry, i didn't know you already had the 2.2.53.3.

for the g10 thing, it's now the way to define the extruder temperature on reprap firmware:
https://duet3d.dozuki.com/Wiki/Gcode#Section_G10_Tool_Offset_and_Temperature_Setting

mafotaso, 2020-11-06:
> sorry, i didn't know you already had the 2.2.53.3.
> 
> for the g10 thing, it's now the way to define the extruder temperature on reprap firmware:
> https://duet3d.dozuki.com/Wiki/Gcode#Section_G10_Tool_Offset_and_Temperature_Setting
> 
> > In others softwares if you select Marlin, it change the speeds configurates when you generate the G-code.
> 
> Go to machines limits and verify that the "enable limit" checkbox is unchecked, so it won't print the limits. Limits can be printed for all firmware (that can accept the gcodes).

dear hero. :)

you have solved the problem :) it is now working fine.

mafotaso, 2020-11-07:
But I justo have discovered a real bug.

If you save a "printer config" in other profile to re-write a profile. Like. "click Save" then "choose a different profile to re-write, and then Super slicer colpases. With this alert:

14:41:45: Internal error: boost::bad_format_string: format-string is ill-formed
14:41:45: Internal error: boost::bad_format_string: format-string is ill-formed

and then Super close itself.

supermerill, 2020-11-08:
i changed a setting in a printer profile, then saved it then select an other printer profile and nothing happened.

closed 2021-06-26 by supermerill