supermerill/SuperSlicer #638: Wipe Into Object Not Working

vertigo235, 2020-11-05:
### Version
Version 2.2.54

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S
### Behavior
Wipe into this object does not appear to be working

#### Project File (.3MF) where problem occurs
Added
[wipe_into_object.zip](https://github.com/supermerill/SuperSlicer/files/5497340/wipe_into_object.zip)

supermerill, 2020-11-06:
What do you mean by "Wipe into this object"?
what makes you believe it's not working? Can you explain me why?

vertigo235, 2020-11-06:
Actually it might be working, but for whatever reason this group of objects just doesn't seem to use much of the waste object, hmm.

vertigo235, 2020-11-06:
Seems to only use it for one color that is on every layer, I guess this is working normally.  

Might just need to close this I guess.

vertigo235, 2020-11-06:
Yeah seems to be a suboptimal behavior I guess, even if you add multiple objects (such as 1 per color that needs to be purged it only seems to use one color per layer into the purge object. Would be nice if it would use a different color on each object to minimize the waste.

vertigo235, 2020-11-06:
Here is a project with 4 purge objects, but it seems to purge the white color in all 4 objects at each layer. 

[wipe_into_object_multiple_purge_objects.zip](https://github.com/supermerill/SuperSlicer/files/5498222/wipe_into_object_multiple_purge_objects.zip)