supermerill/SuperSlicer #639: Defect: Slicing does not work, stuck in negative flow message.

kursatu, 2020-11-06:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
Latest release
### Operating system type + version
_What OS are you using (and version)_
Windows
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
[image]

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_
[Defect13.zip](https://github.com/supermerill/SuperSlicer/files/5499117/Defect13.zip)

supermerill, 2020-11-06:
I computed the wipe tower brim in an unsafe way.
I'll fix that.
in the meantime, you can set a width for the the first layer instead of 0.

kursatu, 2020-11-15:
The attached .3mf file did not show the reported problem, pulled and built with the latest sources on github.

closed 2020-11-15 by kursatu