supermerill/SuperSlicer #640: External Perimeter produces slight overhang on rounded surfaces (?)

thierryzoller, 2020-11-06:
### Version
Latest

Airing on the side of caution an reporting this.
[Wave Square.zip](https://github.com/supermerill/SuperSlicer/files/5502151/Wave.Square.zip)

**TOP View**
[image]

**BOTTOM VIEW**
[image]

supermerill, 2020-11-06:
change 'thin wall' -> 'min width' to 100% and it should go away.

thierryzoller, 2020-11-06:
It's better but not gone 
[image]

supermerill, 2020-11-08:
I set it to 100 and i can see the change.

thierryzoller, 2020-11-09:
I made you a video of it not working 

[b3cf0f82-8494-4269-af21-14fed9b4d562.zip](https://github.com/supermerill/SuperSlicer/files/5510438/b3cf0f82-8494-4269-af21-14fed9b4d562.zip)

thierryzoller, 2020-11-09:
> I set it to 100% and i can see the change.

I downloaded your example and still see the overalp, I believe you might be looking at another place.

supermerill, 2020-11-09:
or maybe you're not at the version i am.

supermerill, 2020-11-09:
my video, on 2.2.54.2:
https://mega.nz/file/d0pSiDyS#SAhv3yuGIlHenV4aIpnMfCAMYITVxHzsmiIG5fhg6LY

thierryzoller, 2020-11-09:
Are there any other settings not included in the 3MF that would somehow influence this behaviour? I can't explain it otherwise

supermerill, 2020-11-09:
all settings are in the .3mf

thierryzoller, 2020-11-09:
What I did in the past was that I overwrote each new SS release with a new one. So I did delete the folder and cleany unziped the release. Problem solved.  It seems that for whatever reason (maybe it didn't overwrite every file, or used an older version) I was running a problematic version.

**Problem solved through clean re-install.**

closed 2020-11-09 by supermerill