supermerill/SuperSlicer #641: Top solid infill not detected

toskium, 2020-11-07:
### Version
SuperSlicer 2.2.54.2

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Klipper 0.90
DIY delta printer

### Behavior
Superslicer does sometimes not detect top solid infill in gcode preview.
[image]

The parts are manifold and I can not spot an obvious reason for its behaviour.

#### Project File (.3MF) where problem occurs
[top_infill_issue.zip](https://github.com/supermerill/SuperSlicer/files/5504599/top_infill_issue.zip)

supermerill, 2020-11-08:
seems fixed in the next 2.3.55 version

toskium, 2020-11-09:
alright, will check on the next version!

closed 2020-11-25 by supermerill