supermerill/SuperSlicer #643: Hotend not heating

nrepian, 2020-11-08:
### Version
2.2.53

### Operating system type + version
Win 10 x64

### 3D printer brand / version + firmware version (if known)
Voron 2.4 and Klipper 0.9.0

### Behavior
Hotend is not heating when g-code is generated using Voron 2.4 0.4mm profiles

Tried with "klipper" and "RepRap" printer configs

#### Project File (.3MF) where problem occurs
[voron_design_cube_v6.zip](https://github.com/supermerill/SuperSlicer/files/5506909/voron_design_cube_v6.zip)

neophyl, 2020-11-08:
Is your printers start gcode correct ?  I don't know what klipper requires as its seems its pretty configurable but the sliced gcode contains
print_start EXTRUDER=240 BED=105
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion

So the question I ask is, are those the correct format for heating your bed and Extruder in your voron/klipper configuration ?

Kirinkunhime, 2020-11-08:
Looks very much like someone does not use the original voron klipper configuration.

supermerill, 2020-11-08:
Also RepRap is using G10, so it isn't compatible with klipper M106.

closed 2022-03-23 by supermerill