supermerill/SuperSlicer #645: Bug in start gcode

Tinchus2009, 2020-11-09:
### Version
2.2.53

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
custom made, reprapfirmware 3.1

### Behavior
My start Gcode looks like this:
T0
M141 S100
M116
G4 S120
M104 S{first_layer_temperature[0] -100}
M116
G28
M400
G29
G29 S1
M141 S{chamber_temperature[0]}
M116
G4 S180
G10 S{first_layer_temperature[0]}
M116

Acording to that, no extra heating commands should be added by superslicer, but that is what is happening. This is the generated gcode:

M107
T0
M141 S100
M116
G4 S120
M104 S150
M116
G28
M400
G29
G29 S1
M141 S80
M116
G4 S180
G10 S250
M116
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
T0
M572 D0 S0.8
G10 S240 P0 ; set temperature
M116 ; wait for temperature to be reached

Those 2 last lines are being added by superslicer, it should be there since I already used a M104 command in the start gcode, and later I used a G10 command to heat up my hotend. Also, supersiler is adding its own G10 command but it is using the "other layers temperature" instead of the first layer temperature.

supermerill, 2020-11-09:
I need your project file

Tinchus2009, 2020-11-09:
I lost the original project but I re created another one showin this behaviour; I have atached the project file and the generated gcode is this one (only the first lines are posted). As you can see the last 2 lines are added by the slicer, and the temperature being used is not the initial layer temp, but the "other layers" temp, 
[pivot_test_pva.zip](https://github.com/supermerill/SuperSlicer/files/5513894/pivot_test_pva.zip)
[pivot_test_pva.zip](https://github.com/supermerill/SuperSlicer/files/5513900/pivot_test_pva.zip)

M107
M141 S100
M116
G4 S120
M104 S150
M116
G28
M400
G29
G29 S1
M141 S80
M116
G4 S180
G10 S250
M116
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
T0
M572 D0 S0.8
T0 P0 ; select tool but don't run any tool change macros
M701 S"/filaments/ABS22_soluble"
T1 P0
M701 S"/filaments/BVOH"

G10 S240 P0 ; set temperature
M116 ; wait for temperature to be reached

supermerill, 2020-11-10:
I fix the bug for the wrong temperature.
I won't remove the useless added lines, because they are here to ensure that everything is correct before printing with a multi-extruder setup. I don't see any drawback, but if you see any, tell me.

closed 2020-11-25 by supermerill

iPeel, 2021-03-25:
Hi, WRT the above comment, I do get issues with this extra M116. As part of my tool change script I have a priming routine, the extra M116 command appears after this and causes a wait which can cause the nozzle to ooze.

supermerill, 2021-06-03:
You can now deactivate all auto-filled start gcode.

SputnikOC3D, 2021-09-04:
> You can now deactivate all auto-filled start gcode.

where is the option to turn off uauto generated start.g code from being appended to my sliced gcode files?

ver . 2.3.56.8  - 

the appended G10 P0 S215 and subsequent M116 is halting my print at the end of my start.g purge line.

supermerill, 2021-09-04:
> where is the option to turn off auto generated start.g code from being appended to my sliced gcode files?

https://user-images.githubusercontent.com/6536403/132107328-69baf790-bbe7-40cf-abdf-3aa5f1b87f77.png