supermerill/SuperSlicer #646: rETRACTION CALIBRATION CODE GENERATOR

old-relic, 2020-11-09:
### Version
2.2.54_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_

### Operating system type + version
_What OS are you using (and version)_
WIN 10 
### 3D printer brand / version + firmware version (if known)
DUET WIFI_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
The first line of gcode that sets the hot end temp is missing the M104 part of the statement
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
[151.388509,68.845262,7.700000],"boundingbox_size":[26.805401,26.806400,15.400000]}
; plater:{"center":[147.804087,142.432591,0.000000],"boundingbox_center":[147.804087,142.432591,7.700000],"boundingbox_size":[198.549114,173.981059,15.400000]}

M107
G90 ; use absolute coordinates
M83 ; extruder relative modeG28 ; home all
 S230 ; set extruder temp
M140 S70 ; set bed temp
M190 S70 ; wait for bed temp

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2020-11-10:
I need the 

> Project File (.3MF) where problem occurs
> Upload a SuperSlicer Project File (.3MF) (File -> Save / Save Project)

old-relic, 2020-11-10:
Your problem reporting web page does not allow .3mf files to be uploaded.

So it is attached to this reply.

supermerill, 2020-11-10:
you can zip it or just renamed it with a .zip extension

old-relic, 2020-11-10:
[retraction_calibration.zip](https://github.com/supermerill/SuperSlicer/files/5519262/retraction_calibration.zip)

old-relic, 2020-11-10:
I’m a noob at driving in GitHub.

I Think I was able to add the zip file.

mike

supermerill, 2020-11-10:
look at your custom gcode, you erase the M104 yourself (line above the M140).

the slicer detects the 109 anyway so it's all green.

old-relic, 2020-11-10:
Thanks for finding that.

Sorry, I should have looked there myself.

Keep Well

Mike

closed 2020-11-10 by supermerill