supermerill/SuperSlicer #647: Tool change not being recogniced?

Tinchus2009, 2020-11-09:
### Version
2.2.53

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
custom made, reprapfirmware
### Behavior
Im setting up a dual extrusion printing profile. My printer has a dual hotend print head, T0 and T1

I have attached the project file. Problem is that my filament start gcode has a sequence of T0 and T1 callings for afilament assignment. I have selected the STL to be printed by T0 (tool 1) and supports by T1 (tool 2). 
The slicer preview shows everuthing like being printed by t1 (tool 2), that is the forst thing I noticed.
And the generated gcode puts the call to T0 in the wrong place: since there is a tool setup in my filaments settings start gcode, the last tool assigned is T1 and not T0 as it should. What seems to be wrong is the place where superslicer puts calls the tool to be used, it should be placed After the filament start gcode so this problem wouldnt occur.
[pivot_test_pva.zip](https://github.com/supermerill/SuperSlicer/files/5513924/pivot_test_pva.zip)
[image]

supermerill, 2020-11-10:
your filament startup is
```
M572 D0 S0.8
T0 P0 ; select tool but don't run any tool change macros
M701 S"/filaments/ABS22_soluble"
T1 P0
M701 S"/filaments/BVOH"
```

so for the slicer, you'll always be on the second extruder after reading the generated gcode.
But the gcode wasn't generated with that in mind.

Imo: you messed with gcode trying to do something and that creates somethign wrong.

Can't fix anything.
If the slicer isn't creating correct TX commands, open an issue on it.

Tinchus2009, 2020-11-10:
Not really, because I have marked the STL to be printed with extruder 1. Despite the STL is marked to be printed with extruder 1, the slicer is treating it as to be printed with extruder 2

supermerill, 2020-11-10:
it begins at 0.
it's only beginning at 1 in the gui, for usability reasons. But if you write code, it begins at 0.

Tinchus2009, 2020-11-10:
Sorry, I didnt understand the answer: in the preview the project I sent you is showing all the print as being printed with extruder2, you mean this is just a preview bug but the gcode geenrated is OK?

supermerill, 2020-11-10:
no, it shows exactly what the gcode is.

if you set the support material extruder to 2, it will change to the second extruder by printing the gcode `T1` before printing support and then it will go back to first extruder by printing `T0`

closed 2021-06-14 by supermerill

Tinchus2009, 2021-06-14:
Ok, this has been closed but the problem remains or at least Im not achieving the desire behaviour. I will check deeper to see if I understand correctly what you meant

supermerill, 2021-06-14:
Maybe I should create a kind of "helpdesk" for the patreons.

[pivot_test_pva_fixed.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6650636/pivot_test_pva_fixed.3mf.zip)