supermerill/SuperSlicer #648: Dual extrusion prints - bug or normal behaviour

Tinchus2009, 2020-11-10:
### Version
latest official

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
custom made reprapfirmare, dual hotend printhead

### Behavior

Im trying to setup a dual material printing (abs + soluble supports)
My problem is the priming tower. I have been trying to activate it, but it always forces me to setup the supports extruoder to a value of cero, and to my understanding that means the supports will be built using the last active extruder, and that wpuld be the ABS extruder in my case.
Is there any way to assign extruder 2 to be used as the extruder using the soluble support material  for printing the supports and activate the priming tower?

supermerill, 2020-11-10:
activate supports, set them to soluble, activate synch with object, go to multiextruders page, activate teh wipe tower if not already here, set 2 to the support extruder.

[multi_soluble.zip](https://github.com/supermerill/SuperSlicer/files/5517464/multi_soluble.zip)

Tinchus2009, 2020-11-10:
Great! thanks for the help. I think there is something to improve there, the plane setting needs to be set to "none", so the popup information is not very clear I think.
I have tried to copy your settings, but when I try to slice, everything is sliced for extruder 2. I have uploaded the project
[multi_soluble_wrong_tool.zip](https://github.com/supermerill/SuperSlicer/files/5518325/multi_soluble_wrong_tool.zip)

supermerill, 2020-11-10:
it's becaus eof your custom fiament gcode that set the extruder to extruder 2 whatever you do.

Tinchus2009, 2020-11-10:
I was suspecting that, but that command just assigns a material to an extruder. Shouldnt the slicer assign by itself the tool acording to my setup? I specifically assigned T1 for the STL

supermerill, 2020-11-10:
yes, the slicer should do it itself. You can see it emit a TX before.
if you need a repeat the command, use variable to get the current extruder: `T[next_extruder]` (you can also get the previous number with previous_extruder)

closed 2021-06-14 by supermerill