supermerill/SuperSlicer #649: Build fails on AUR due to missing .desktop file

thecakeisalie25, 2020-11-10:
### Version
2.2.54.2.r5.gf7bf94840-1 (superslicer-git on the aur)

### Operating system type + version
Arch Linux Rolling

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Build fails on aur package `superslicer-git` with the following error:
```
CMake Error at cmake_install.cmake:66 (file):
  file INSTALL cannot find
  "/home/typecasto/.cache/yay/superslicer-git/src/SuperSlicer/src/platform/unix/SuperSlicer.desktop":
  No such file or directory.
```
I was able to fix the issue on my local machine by writing a shell script that creates the file every 5 seconds, and running that while building, but that's less than ideal. This should be a fairly easy issue to fix for anyone who knows how .desktop files work. Just edit the /src/platform/unix/PrusaSlicer.desktop file and rename it to SuperSlicer.desktop, and it should work.

#### Project File (.3MF) where problem occurs
N/A

supermerill, 2020-11-10:
Yhanks for the report, I'll fix it.
you have to rename the Prusaslicer.desktop to SuperSlicer.desktop in .\src\platform\unix

Lukas1818, 2020-11-17:
I test it with commit b686b8956 and it does built, without any issues.
But there is another [issue](https://github.com/supermerill/SuperSlicer/issues/676) at runtime now.

May I should remove the .desktop file from the AUR and use the .desktop, which is included in the SuperSlicer repo now.

closed 2020-11-25 by supermerill