supermerill/SuperSlicer #650: Brim failing

Tinchus2009, 2020-11-10:
### Version
latest stable

### Operating system type + version
win

### 3D printer brand / version + firmware version (if known)
custom made coreXY
### Behavior
I have been printing tests and leanring supersilcer new features, definitly this is one of the best finds I have done in terms of slicing, super thanks for this great version.
During tests, I print a lot of ABS, im suffering of warpings and I have just noticed the reason. Not exactly in the example Im showing in the capture because it is just a test of my new dual material profile, but it shows the problem: when you have a model that needs supports, regardless if it is soluble or not, when you activate the brim to help with the warping, if there is support around the object, the brim is not created around the object but around the support, and makes the brim useless.
The proper behaviour I think would be the brim to be created and support to be createn on the second layer on top of the brim.

[image]

supermerill, 2020-11-10:
[image]

Tinchus2009, 2020-11-11:
I tried that option, but the result is a solid initial layer but not conected to the object. Look at this picture, it has that option activated. The red square i the center is the object, then you can see the support, solid layer in green and then the brim surrounding the support layer. And even with this option, take a lok at the separation between the solid layer and the object.
[image]

supermerill, 2020-11-11:
oh, right.

bung69, 2020-11-18:
A hacky temporary fix is to use the "Brim Ears" option with the max angle set to 180deg

However this causes the support's raft to be printed inside the brim which could cause a bit of a mess / nozzle build up.

IMO the raft under support should just be out right disabled and instead allow independently configurable top and bottom support interface layers.  

[image]

closed 2021-12-22 by supermerill