supermerill/SuperSlicer #652: RepRap GCode "wrong order"

peterthepeter, 2020-11-10:
### Version
2.2.54

### Operating system type + version
macOS 10.15.7

### 3D printer brand / version + firmware version (if known)
Vividino Troodon

### Behavior
Super Slicer creates in the gCode file the commands for 
Set  Bed to Temperature and wait until is arrived 
Set Standby Temperatur (Hotend)
Set Temperatur (Hotend)

but after this my normal Startcode "will start" but I think after this a M116 is needed so the Hotend can reach the Set Temperature. For now the Printer continue the gCode and does not wait for the Hotend.
Because so the Bed is a Temperature the Hotend Heat Up and the Printer starts the "custom gCode" like Level Bed etc. and then the in my case the Printer wants to Print a Purge Line but the Hotend is to cold.

I know normally I preheat everything, but I think with a M116 after the 3 line were a first world Problem solved ;).

; generated by SuperSlicer 2.2.54 on 2020-11-10 at 22:55:25 UTC

; 

; external perimeters extrusion width = 0.42mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.42mm
; first layer extrusion width = 0.45mm

M107
M190 S60 ; set bed temperature and wait for it to be reached
G10 P0 R200 S200 ; sets the standby temperature
G10 P0 S200 ; set temperature   
M42 P24 S0
M42 P26 S0
M42 P25 S255
T0
G92 E0
M42 P25 S0
M42 P26 S0
M42 P24 S255
G32
G29 S1
G92 E0
G1 X40 Y0 F4800
G92 E0
G1 E-3.0000 F1800
G1 Z0.200 F1000
M570 P0 P30 T10
G1 X200 Y0 E30.0000 F1000
G1 E-1.0000 F5400
M570 H1 P20 T10
M572 D0 S0.04
G10 P0 S200 ; set temperature
M116 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0

supermerill, 2020-11-11:
I don't have your project file, so it's a guess:
I can't infer on what is needed in the custom_start_gcode. If you need the extruder at a temp, please put the needed gcode in the custom_start_gcode.
If i put a wait before it, i will have a shitton of complain by people that don't want to wait at this moment.

peterthepeter, 2020-11-11:
I only thing I want is that the gCode wait Until the Extruder Temp is reached. The Slicer generate the first 4 Lines (there are not from the start gcode).

M107
M190 S60 ; set bed temperature and wait for it to be reached <- here you wait for the Temperature to wait until is it reached
G10 P0 R200 S200 ; sets the standby temperature
G10 P0 S200 ; set temperature <- here so set only the temperature but you don't wait. If you ad here after the Line an M116 the extruder wait.

Okay why people would noch wait until the temperature of the Extruder is reached bevor everything else is started?

These are just my 2 cents here.

neophyl, 2020-11-11:
If you want to wait then add that line into YOUR custom start gcode.  That way everyone gets what they want.  The whole point of those sections is to make it work like the end user wants.
If you add things in 'just because it should work this way' then most of the time there will be someone with a use case the opposite so leaving it up to the user is the best way.

supermerill, 2020-11-11:
I don't have your project file, i can't help more

supermerill, 2021-06-01:
now you can disable the auto-start gcode

closed 2021-06-01 by supermerill