supermerill/SuperSlicer #653: Build failure on WIndows

kursatu, 2020-11-11:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
Pulled the sources tp the latest , ie. commit a44039c3

### Operating system type + version
_What OS are you using (and version)_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
I was building before this pull.
Followed the steps https://github.com/supermerill/SuperSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md#how-to-build
- The build always fails with the following
       "C:\Users\kursa\Documents\GitHub\SuperSlicer.1\build\INSTALL.vcxproj" (default target) (1) ->
       (PostBuildEvent target) ->
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.ta
       rgets(153,5): error MSB3073: The command "setlocal [C:\Users\kursa\Documents\GitHub\SuperSlicer.1\build\INSTALL.
       vcxproj]

There is also an error in the documentation. The line should read this, otherwise cmake does not run.
cmake .. -G "Visual Studio 16 2019" -A x64 -DCMAKE_PREFIX_PATH="PATH_TO_SuperSlicer\deps\build\destdir\usr\local"

After  these failure, the VS project is not usable, it just doe not build. All the subprojects are marked as incompatible.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2020-11-11:
thanks you for the fix for the doc.
I have no problem building it with widows10x64 from a clean fresh clone, on my computer and in the continous integration server.

supermerill, 2020-11-12:
oh, it seems that the windows build has a problem and create a SuperSlicer.dll instead of a slic3r.dll, you may have to rename it until i correct it.

kursatu, 2020-11-12:
I've created a symlink , otherwise development would be hard.
mklink slic3r.dll SuperSlicer.dll

kursatu, 2020-11-15:
Closing this, there is a workaround and another defect filed for it.

closed 2020-11-15 by kursatu

supermerill, 2020-11-15:
now it should use the SuperSlicer.dll, doesn't need to rename.