supermerill/SuperSlicer #655: Unable to locate in SuperSlicer where to put gcode for purge bucket for multiprinting.

fatdaddypaul, 2020-11-11:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_

### Operating system type + version
_What OS are you using (and version)_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2020-11-11:
* filament ->end_filament_gcode is printed
* printer -> custom gocde -> toolchange_gcode is printed, 
* the `TX` is printed
* the filament ->start_filament_gcode is printed
[in custom gocde, you can access the special variables](https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language) next_extruder, previous_extruder, * layer_num, layer_z and all the normal ones

fatdaddypaul, 2020-11-12:
I wrote this code but missing something =, it retracts the previous_extruder, but does not prime the next_extruder.
[previous_extruder]
M104 T[previous_extruder] S[first_layer_temperature] ;set hotend temperature
G91 ; Switch to relative positioning
G1 E-5 F600
G1 Z10 F1000 ; Lift nozzle by 10 mm
G90 ; Return to absolute positioning
G1 X-7 F6000
G1 E-60 F300; Retract 65 mm
G92 E0 ; Reset extrusion distance on old tool

[next_extruder]
M104 T[next_extruder] S[first_layer_temperature] ;set hotend temperature
G1 X-7 F6000 ; Go to X=-7
G92 E0 ; Reset extrusion distance on new tool
G1 E105 F300 ; Extrude 100 mm
G92 E0 ; Set current extruder position to 0mm
G0 F10000 ; Increase feed rate before next travel.
G1 E-1 F1000

supermerill, 2020-11-12:
your [next_extruder] gcode isn't printed?

fatdaddypaul, 2020-11-12:
My code when sliced produces this in the gcode..
M104 T-1 S215 ;set hotend temperature
G91 ; Switch to relative positioning
G1 E-5 F600
M73 P0 R78
G1 Z10 F1000 ; Lift nozzle by 10 mm
G90 ; Return to absolute positioning
G1 X-7 F6000
G1 E-60 F300; Retract 65 mm
G92 E0 ; Reset extrusion distance on old tool

0
M104 T0 S215 ;set hotend temperature
G1 X-7 F6000 ; Go to X=-7
G92 E0 ; Reset extrusion distance on new tool
G1 E105 F300 ; Extrude 100 mm
G92 E0 ; Set current extruder position to 0mm
G0 F10000 ; Increase feed rate before next travel.
G1 E-1 F1000

fatdaddypaul, 2020-11-12:
My extruders are T0 and T1.

supermerill, 2020-11-12:
to avoid to write the custom gcode whne the exruder isn't here yet (-1) if the if:

```
{if previous_extruder >= 0}
M104 T[previous_extruder] S[first_layer_temperature] ;set hotend temperature
G91 ; Switch to relative positioning
G1 E-5 F600
G1 Z10 F1000 ; Lift nozzle by 10 mm
G90 ; Return to absolute positioning
G1 X-7 F6000
G1 E-60 F300; Retract 65 mm
G92 E0 ; Reset extrusion distance on old tool
{endif}
```

There is something else that is not right? (if yes, please write the expected gcode vs the one you have nd your custom gcode, or even better: your zipped project file)

closed 2021-06-04 by supermerill