supermerill/SuperSlicer #659: External perimeters cutting off

TD5023, 2020-11-11:
### Version
2.3.55

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
In the newest update, there are points where the external perimeter does not connect, leaving gaps to the internal perimeter. As far as I can see, all of the internal perimeters complete full loops as expected. I have not done a print, but loading into a separate gcode viewer shows that it is definitely a code issue and not just a visual error in SS. I'm not sure if it's just a setting issue on my end, but since I imported an old working config file and only adjusted the new settings, I don't think that's the case. It actually appears that instead of completing the loop, the extruder doubles back over itself.

[image]
[image]
[image]

#### Project File (.3MF) where problem occurs
[external perimeter issue.zip](https://github.com/supermerill/SuperSlicer/files/5527353/external.perimeter.issue.zip)

supermerill, 2020-11-12:
That's the cooling fan speed-up time, that cut the perimeter in a wrong way. set it to 0 to deactivate it.

TD5023, 2020-11-12:
Thanks! That fixed the problem for me since I don't care about running a different fan speed on the outer layers. Wouldn't this be a bug for the people who do, though?

closed 2020-11-18 by TD5023