supermerill/SuperSlicer #661: BUG Support incorrectness

karabas2011, 2020-11-12:
### Version
2.2.54

### Operating system type + version
Win8

### 3D printer brand / version + firmware version (if known)
marlin

### Behavior
Support bridges no matter  on or off "Dont support bridges"
Support goes through walls of model !!!

#### Project File (.3MF) where problem occurs
[image]
[image]

_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

karabas2011, 2020-11-12:
[ss.zip](https://github.com/supermerill/SuperSlicer/files/5529534/ss.zip)

supermerill, 2020-11-12:
You may want to increase the XY separation a tad.

karabas2011, 2020-11-12:
5mm
[image]

supermerill, 2020-11-12:
There are overhangs, and these are not considered as bridges, i think that's your problem. Ideally, it should detect that your overhangs are strait and so should be considered as bridges, but there is nothing in the software currently for that.

OS if it's what you ask for, it's a new feature/improvement to create.

closed 2020-11-25 by supermerill