supermerill/SuperSlicer #662: V.2.2 - Settings and Printers Disappear

thierryzoller, 2020-11-12:
### Version
2.2 Stable

Status:
- I have prusaslicer installed
-

What is happening:
- When I install the new 2.2 stable, some settings error (can't read monotonous) and Printers dissapear. Downgrading to repetier-version, fixes the issue.

neophyl, 2020-11-12:
Cant comment on the 2.2 stable as I haven't downloaded it yet but....

The 2.3.55 I think is in the same boat as the Prusa 2.3.0 alpha releases.  They save all their configs to an alpha directory so that it doesn't impact the configurations of the vanilla released versions.
If you want any custom profiles from 2.2.xx in the 2.3.xx versions then you can export a configuration bundle from the earlier release and then import it into the 2.3.xx release.  That's normal for proper releases and alpha/beta builds.

I did run the 2.3.55 and ran through the wizard to get a basic mk3 installed and the software running and then I imported a saved config bundle I had from 2.2 and it brought through all my custom Print/Filament/Printer profiles.  There was no problems importing those, although I don't use octoprint so didn't have to deal with the new 'Physical' printer changes that have been added to the PS 2.3 alphas that 2.3.55 is merging with.

supermerill, 2020-11-12:
You can't really downgrade from the 2.3/2.2.54 to the 2.2.53 without these problems, I have to add a dialog to ask for erasing incompatible settings.

For the second problem, it's as neophyl said.

justdanpo, 2021-02-21:
I've lost my print and filament settings after running 2.3.55. I've tried restoring settings from backup but that hasn't helped. Are there other settings except SuperSlicer folder in a user profile?

supermerill, 2021-06-02:
the 2.3.55 branch uses the `superslicer-alpha` directory (because it was using the alpha-then-beta prusaslicer 2.3). Reverted back to `superslicer` in 2.3.56

closed 2021-06-02 by supermerill