supermerill/SuperSlicer #663: translation into Russian language

joker13joker, 2020-11-12:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_

### Operating system type + version
_What OS are you using (and version)_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_
hi I would like to know about the possibility of translation into Russian? I can help with the translation.

supermerill, 2020-11-12:
You can follow this guide
https://github.com/supermerill/SuperSlicer/wiki/Creating---updating-a-translation
if there are problems, errors or inconsistency with it, tell me.

VectorPort, 2020-11-13:
[ru.zip](https://github.com/supermerill/SuperSlicer/files/5538924/ru.zip)
Папку ru расположить по пути resources\localization. Перевод должен сразу подхватиться, если нет то включить в программы.
Вот перевод на Русский. Пока довольно сыроватый, но переведено вроде все, много где остался машинный перевод, но основные настройки постарался перевести как положено.
Перевод был сделан для 2.2.53 версии, для версии 2.3.55 пока перевод не делал, там много нового текста и часть текста уплыла.

userosos, 2020-11-14:
> 
> 
> [ru.zip](https://github.com/supermerill/SuperSlicer/files/5538924/ru.zip)
> Папку ru расположить по пути resources\localization. Перевод должен сразу подхватиться, если нет то включить в программе.
> Вот перевод на Русский. Пока довольно сыроватый, но переведено вроде все, много где остался машинный перевод, но основные настройки постарался перевести как положено.
> Перевод был сделан для 2.2.53 версии, для версии 2.3.55 пока перевод не делал, там много нового текста и часть текста уплыла.
> 
> Place the ru folder along the resources \ localization path. The translation should be picked up immediately, if not, then include it in the programs.
> Here is the translation into Russian. So far, it's rather damp, but everything seems to have been translated, machine translation has remained a lot, but we tried to translate the main settings as expected.
> The translation was made for version 2.2.53, for version 2.3.55 I have not done the translation yet, there is a lot of new text and some of the text has left.

Good! I want help to you. I think that we need use web platforms for translation like https://www.transifex.com/ for example.

VectorPort, 2020-11-14:
> Good! I want help to you. I think that we need use web platforms for translation like https://www.transifex.com/ for example.

Попробую разобраться с сайтом, но он почему то не хочет добавлять текущий перевод. Он предлагает делать мне перевод с русского на русский, или полностью с нуля на русский.

----
I'll try to deal with the site, but for some reason it doesn't want to add the current translation. He offers to do me the translation from Russian into Russian, or entirely from scratch in Russian.

VectorPort, 2020-11-14:
Все желающие могут присоединиться к переводу на Русский язык. Сделаем SuperSlicer еще популярнее!

[www.transifex.com/superslicer_ru/](url)

joker13joker, 2020-11-15:
> Все желающие могут присоединиться к переводу на Русский язык. Сделаем SuperSlicer еще популярнее!
> 
> [www.transifex.com/superslicer_ru/](http://www.transifex.com/superslicer_ru/)

не работает ссылка.
the link doesn't work.

VectorPort, 2020-11-15:
> > Все желающие могут присоединиться к переводу на Русский язык. Сделаем SuperSlicer еще популярнее!
> > [www.transifex.com/superslicer_ru/](http://www.transifex.com/superslicer_ru/)
> 
> не работает ссылка.
> the link doesn't work.

Fixed

userosos, 2020-12-02:
@supermerill Can you send to us an file for translate Last version SS from Eng to Rus language, please?

supermerill, 2020-12-02:
I need to create a new software to add the strings from the UI files, or put them in the c++code instead.

userosos, 2020-12-02:
> I need to create a new software to add the strings from the UI files, or put them in the c++code instead.

Hmm. Can't we translate SS into other languages now?
Approximate terms of creating this utility?

VectorPort, 2020-12-16:
@supermerill hello.
Completed full translation for version 2.3.55.2. He translated the whole text. I had to write a lot of text myself into the code for translation. As a new translation is released, will I upload new versions here, or what is the best way to send them?
--------
Закончили полный перевод для версии 2.3.55.2. Перевел вроде весь текст. Пришлось много текста самому прописывать в код для перевода. По мере выхода нового перевода буду сюда загружать новые версии, или как лучше их отправлять?
[ru.zip](https://github.com/supermerill/SuperSlicer/files/5702477/ru.zip)

djperya, 2021-01-03:
Присмотритесь к моему переводу, многий текст msgid  вылавливал с dll https://github.com/djperya/localization-SuperSlicer/tree/master/ru

VectorPort, 2021-09-20:
https://github.com/supermerill/SuperSlicer/discussions/1579