supermerill/SuperSlicer #664: Latest nightly won't start

1N4148, 2020-11-12:
### Version
Latest nightlies

### Operating system type + version
Windows 10 H2 64bit

### 3D printer brand / version + firmware version (if known)
Creality Ender 3/5

### Behavior
`create settings  print.ui
No overhangs_width_speed in ConfigOptionsGroup config, tab print.ui.
No seam_angle_cost in ConfigOptionsGroup config, tab print.ui.
No seam_travel_cost in ConfigOptionsGroup config, tab print.ui.
No hole_size_threshold in ConfigOptionsGroup config, tab print.ui.
No infill_connection in ConfigOptionsGroup config, tab print.ui.
No ironing in ConfigOptionsGroup config, tab print.ui.
No ironing_type in ConfigOptionsGroup config, tab print.ui.`

supermerill, 2020-11-12:
delete slic3r.dll, rename superslicer.dll to slic3r.dll

1N4148, 2020-11-12:
Ok, now it's working 👍

kursatu, 2020-11-12:
I've created a symlink via the command below, otherwise development would be hard.
mklink slic3r.dll SuperSlicer.dll

kursatu, 2020-11-15:
I've verified and found this to be working, no need to rename nor create a symlink.
This is done by pulling and building the latest sources on GitHub.

closed 2020-11-25 by supermerill