supermerill/SuperSlicer #666: Start programm

epson950, 2020-11-13:
### Version
2.2.54

### Operating system type + version
Windows 10 64

### 3D printer brand / version + firmware version (if known)
Sovol SV01

### Behavior
When you start a program through a superslicer_local-settings.bat file, it starts, and if you run it through superslicer.exe or superslicer_console.exe, nothing happens and the program does not work.

supermerill, 2020-11-14:
you compile it yourself / downloaded the nightly?
if yes, there is currently a compile bug i have to resolve, you have so delete the slic3r.ddl and rename the superslicer.dll to slic3r.dll

supermerill, 2020-11-14:
that issue is hellish btw

epson950, 2020-11-14:
I'll probably wait for the next version. I noticed another problem - after sending the model to print, it turned out to be 10% more on X and Y

supermerill, 2020-11-14:
maybe your filament shrink compensation?

epson950, 2020-11-14:
Yes, it's filament shrink compensation! I'm sorry, I don't know English well, so these are my mistakes.
And on the first question - I downloaded the night release, but I could not fix the launch, the file superslicer.dll is not available!

commit c6623bb2 (2018-07-30): Increased camera theta max to 180 degrees. Fixes #666
closed 2020-11-25 by supermerill