supermerill/SuperSlicer #667: Z-Seam position settings (Question)

LordMelkor, 2020-11-13:
### Version
2.3.55

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
PRUSA i3 MK3S

### Behavior
[image]

Sharpest corner is no longer an option for z-seam position. How can I replicate that functionality with the "Cost-based" parameters?

supermerill, 2020-11-14:
You have to use 100% angle cost and 20% travel cost.

if you want earest, you should use 100% angle cost and 100% travel cost

closed 2021-06-04 by supermerill

Really01101001, 2023-04-09:
"cost" based z seam location? Seriously? What's wrong with user defined coordinates location for z seam/layer start location? Just trying to make producing excellent prints harder than it needs to be?