supermerill/SuperSlicer #669: Settings are not emitted at the end of the gcode file, which prevents post processor functions

kursatu, 2020-11-15:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
I've built it from the latest sources on GitHub, so it is 2.2.55

### Operating system type + version
_What OS are you using (and version)_
WIndows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
I've a somewhat complicated post processor that relies on the settings. 
They are usually stored at the end of the gcode file
Things like 
; infill_speed = 80
; external_perimeter_speed = 30

They are not emitted anymore, so my post processor falls back to hard-coded default settings.
This does not really work.
Not sure if this is Prusa merge issue or not.
[Defect14.zip](https://github.com/supermerill/SuperSlicer/files/5541769/Defect14.zip)

* _Steps needed to reproduce the problem_
- Slice and save the attached 3mf file.
- Observe that there are no settings at the end of the gcode file.

* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2020-11-15:
solution: update to 2.3.55_hotfix version
you cana lso get the latest nightly

kursatu, 2020-11-15:
I've built the latest sources and the problem is gone.

closed 2020-11-15 by kursatu