supermerill/SuperSlicer #670: Traslation into Spanish Language

jccf07, 2020-11-15:
### Version
2.2.53

### Operating system type + version

Xubuntu 19

### 3D printer brand / version + firmware version (if known)
Ender 3 + Klipper

### Behavior

The Spanish language has some fields that are not translated, I offer myself to translate everything that is not translated

Do you have a text file with all the commands? I could translate from English and then share in github

jccf07, 2020-11-15:
I found this:

https://github.com/supermerill/SuperSlicer/blob/master/doc/Localization_guide.md

I´ll try in a few days, any help it is wellcome

supermerill, 2020-11-15:
if you can't compile the .pot, the latest nightly should have an up-to-date one