supermerill/SuperSlicer #671: Filament/Print speed override is regressed

kursatu, 2020-11-16:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
The latest source from the github. Up to date to commit 47886f24

### Operating system type + version
_What OS are you using (and version)_
Windows 10 
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
The filament speed override seems to be regressed. I was relying on this to limit 
tricky filaments, but noticed that it does not override the speed anymore.

* _Steps needed to reproduce the problem_
Slice the attached .3mf file
Observe that the speed it not 5mm/sec, choosen to show that it is not working.

  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_
[Defect16.zip](https://github.com/supermerill/SuperSlicer/files/5543590/Defect16.zip)

supermerill, 2020-11-16:
can't reproduce, can you send your project file?

1N4148, 2020-11-16:
Caution: 5mm³/s isn't 5mm/s. 5mm³/s may result in 70-80mnm/s based on your line height & width!

kursatu, 2020-11-16:
It was a mistake on my end, was looking incorrect extruder. There is no defect here. Closing.

closed 2020-11-16 by kursatu

kursatu, 2020-11-16:
@1N4148 The unit for the speed is wrong, 
This is another defect. it should have read mm/sec. 
The unit for the volumetric speed is right, which is mm^3/sec.
https://user-images.githubusercontent.com/1686475/99204019-27efac00-2769-11eb-8d63-306af14c1670.png