supermerill/SuperSlicer #672: Debug build fails to start from within VS

kursatu, 2020-11-16:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
Latest code from github, built on my machine.

### Operating system type + version
_What OS are you using (and version)_
Windows 10
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
Follow the steps https://github.com/supermerill/SuperSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md
Open VS and press F5
Observe the error
SuperSlicer.dll was not loaded, error code: 126
Is the build from within VS is reliable?
Or what is the recommended way of building for development purposes?
Maybe this is also a documentation issue, due to build changes.
On a related note, attempt to build from command line fails, probably because of insufficient 
privileges', but it generates the .dll in such a way that it makes the VS Debug work.
msbuild /m /P:Configuration=Debug INSTALL.vcxproj

         -- Installing: C:/Program Files (x86)/SuperSlicer/resources
         CMake Error at cmake_install.cmake:44 (file):
           file INSTALL cannot make directory "C:/Program Files
           (x86)/SuperSlicer/resources": No such file or directory.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2020-11-16:
Did you have libgmp-10.dll and libmpfr-4.dll near your .exe ?

kursatu, 2020-11-16:
Now those 2  dll's are there. don't know if they were missing earlier.
This was a clean build, git cloned first, then build using the steps in the documentation.
Then I tried to debug from within VS, it then failed. This used to work.
Then built with Debug from command line, ignored the errors, VS started to work.
It might be that the VS build does not generate those dll's.

kursatu, 2020-11-16:
@supermerill , thanks for the responses to this and all the other recent defects.
Your work on this is great. It looks like  quite a bit of time commitment.

closed 2020-11-16 by kursatu