supermerill/SuperSlicer #674: Standby temperatures

Tinchus2009, 2020-11-17:
### Version
latest stable 2.2.53

### Operating system type + version
win10
### 3D printer brand / version + firmware version (if known)
custom corexy reprapfirmeare

### Behavior
I have just downloaded latest version and generated a code (calibration of offsets)
And this behaviour just happened: there a a standby temperatures added to the begining of the gcde, and I dont know where they are coming from

G10 P0 R200 S240 ; sets the standby temperature
G10 P1 R200 S240 ; sets the standby temperature

[nozzle_vernier_fixed_large.zip](https://github.com/supermerill/SuperSlicer/files/5552462/nozzle_vernier_fixed_large.zip)

[test_calibration.zip](https://github.com/supermerill/SuperSlicer/files/5552468/test_calibration.zip)

supermerill, 2020-11-17:
that's because you're using reprapfirmware and have a multi-extruder setup.
something is wrong with these gcodes?

Tinchus2009, 2020-11-18:
ok. then it might be another problem, if you take a look at the project file: toolchange temperature enabled was not activated, so I though those configuration option wouldnt affect the gcode.

supermerill, 2020-11-18:
yes, now that switch is taken in account

closed 2020-11-25 by supermerill

bsjoelund, 2022-10-20:
I can not get rid of this  line   G10 P0 R225
W10
SS 2.5.59
Reprap   Only one toolhead T1