supermerill/SuperSlicer #675: Q modified gcode at the beginning ?

balloni55, 2020-11-17:
### Version
2.2.53

### Operating system type + version
WIN10 64

### 3D printer brand / version + firmware version (if known)
Anycubic I3 mega, Marlin 1.1.9

### Behavior
I have only been using SuperSlicer for a short time, so sorry

10 days ago i got another gcode when i slice the identical .stl , this process i like better and want it back.
Unfortunately I could not discover what I have changed by mistake.

thank you for the help
[image]

supermerill, 2020-11-17:
seems like you changed your start_gcode (printer-> custom gcode -> start)

balloni55, 2020-11-17:
i think it´s default, isn´t it?
[image]

supermerill, 2020-11-17:
if you want your gcode to start with 
```
M190 S65
M109 S200
G28
G1 Z5 F5000
```
you can copy-paste
```
M190 S[first_layer_bed_temperature] ; set and wait for bed temp
M109 S[first_layer_temperature] ; set and wait for extruder temp
G28 ; home all axis
G1 Z5 F5000
```

balloni55, 2020-11-17:
Thanks for your help, this work

closed 2021-06-04 by supermerill