supermerill/SuperSlicer #676: crash aftern open the Printsettings.

Lukas1818, 2020-11-17:
### Version
2.3.55.r19.g[b686b8956](https://github.com/supermerill/SuperSlicer/commit/b686b8956772de4f039e00b65b8caf8dea6fc636)
AUR: [superslicer-git](https://aur.archlinux.org/packages/superslicer-git/)

### Operating system type + version
Manjaro Linux (rolling release)

### Behavior
Crash if I open the Printsettings. A error popup appears:

[image]

supermerill, 2020-11-17:
maybe an incompatible version of gtk / wxwidget ?
Can't do anything on my side.
You can also try to build it without asserts (in release mode)...

Lukas1818, 2020-11-17:
I use wxgtk3 3.1.4 and gtk3 3.24.23.

The issue do not happen with SuperSlicer 2.2.54.2.

supermerill, 2020-11-17:
This is more probably because of the new way prusaslicer display the setting labels: by painting them directly on the screen instead of using the text widget.

I don't know why it complain on your os. In release mode, on ubuntu 18.04, with the CI build script, there is no issue i know.

msrd0, 2020-11-17:
I can reproduce this on ArchLinux (rolling release / 5.9.8-arch1-1) running sway, both native wayland and through xwayland.

msrd0, 2020-11-17:
Testing prusa-slicer-git, this doesn't seem to come from upstream changes, as their print tab opens perfectly fine out of the box.

supermerill, 2020-11-17:
the backtrace doesn't go father away, in the slicer methods?

msrd0, 2020-11-17:
their current master, so some commits after 2.3-alpha3

here's the complete backtrace:

```
ASSERT INFO:
./src/gtk/dc.cpp(343): assert ""cr"" failed in wxPaintDCImpl(): using wxPaintDC without being in a native paint event

BACKTRACE:
[1] wxPaintDC::wxPaintDC(wxWindow*)
[2] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[3] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[4] wxEvtHandler::TryHereOnly(wxEvent&)
[5] wxEvtHandler::ProcessEventLocally(wxEvent&)
[6] wxEvtHandler::ProcessEvent(wxEvent&)
[7] wxNotebookBase::SendPageChangedEvent(int, int)
[8] g_closure_invoke
[9] g_signal_emit_valist
[10] g_signal_emit
[11] g_closure_invoke
[12] g_signal_emit_valist
[13] g_signal_emit
[14] gtk_main_do_event
[15] g_main_context_dispatch
[16] g_main_loop_run
[17] gtk_main
[18] wxGUIEventLoop::DoRun()
[19] wxEventLoopBase::Run()
[20] wxAppConsoleBase::MainLoop()
[21] wxEntry(int&, wchar_t**)
[22] __libc_start_main

supermerill, 2020-11-17:
I don't have a clue of what can cause this...

Can you compile the prusa source at the ea5fdcd7b1e5e26e105f078ff78dfb4c18bd7743 commit? just to be sure it's not an already fixed bug.

msrd0, 2020-11-17:
I can duplicate the bug with ea5fdcd, so this is probably a fixed bug.

supermerill, 2020-11-17:
should be fixed in the next merge.

Lukas1818, 2020-11-23:
The crash does not happen at version adf6f72a1b0f99f82efe689f3f28bd74eae5caaf anymore.

closed 2020-11-23 by Lukas1818