supermerill/SuperSlicer #678: SuperSlicer uses smaller area for bridges than PrusaSlicer

carloshespanhol, 2020-11-17:
### Version
2.3.55

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Hypercube - Klipper

### Behavior
SuperSlicer uses smaller area for bridges than PrusaSlicer 

See images attached

[image]

[image]

supermerill, 2020-11-17:
set infill -> advanced -> anchor solid infill by Xmm -> bridged to 3 to replicate prusa values

carloshespanhol, 2020-11-18:
In a simple bridge it works, but not in this case. I attached the stl file.

What I find confusing is that you said to change "anchor solid infill"  but the color is of bridge and not solid infill, it's not intuitive.

stl file compacted
[hole_bridge.zip](https://github.com/supermerill/SuperSlicer/files/5560115/hole_bridge.zip)

supermerill, 2020-11-18:
can't reproduce
[image]

maybe export your project fie?

carloshespanhol, 2020-11-18:
I found my error. I have set "No perimeters on bridge area" just disabled it and worked.
Thanks for your help.

Hywelmartin, 2020-12-09:
I can't get the "slab" back in my part..
[image]

[image]

[image]

supermerill, 2020-12-09:
infill -> anchor -> bridge

Hywelmartin, 2020-12-10:
Doesn't work.. there must be some setting I'm screwing up.. cant get rid of the perimiters around the "slab"
[magic_bridge.zip](https://github.com/supermerill/SuperSlicer/files/5674797/magic_bridge.zip)

supermerill, 2021-04-27:
maybe because of the buggy 'one peri on top' that plague the 2.3.55
reopen if the error persist.

closed 2021-04-27 by supermerill