supermerill/SuperSlicer #679: BRidge/Overhand detection seems to be regressed

kursatu, 2020-11-17:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
Both Alpha 2.3 and the build from the latest sources from github.

### Operating system type + version
_What OS are you using (and version)_
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Vking customized

### Behavior
* _Describe the problem_
The bridge and overhang detection seems to be regressed.
* _Steps needed to reproduce the problem_
Slice the attached 3mf file
Observe that both bridge and overhangs are not detected.

* _Expected Results_
Both overhangs and bridge should be displayed as such.
* _Actual Results_
They are shown as regular perimeters.
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]

#### Project File (.3MF) where problem occurs
[Defect17.zip](https://github.com/supermerill/SuperSlicer/files/5555869/Defect17.zip)

supermerill, 2020-11-17:
You have the soluble supports activated, this combination remove the overhangs detection.

I wonder by how much I should alter this behavior.

kursatu, 2020-11-17:
Thanks for the quick response and good work towards making the superslicer better.
It should **not disable** 2 essential features w/o user control, and maybe give option(s) elsewhere to disable them.

I have been using polymaker/polysupport filament as soluble interface (ie. no-gap) support for years.
I also print complicated functional parts with holes in them so that they can be assembled with other parts
or with other reinforcement materials.
They require overhang/bridge detection with dimensionally accurate extrusion (ie. no over extrusion)
and high quality , easy to detach supports.
This current behavior would make the superslicer simply unusable for me. 
I'll, of course will change the code, if I need to, which I already did sometime ago for overhang 
and bridge over extrusion.

The current behavior is problematic at many aspects.
It is inconsistent, in the sense that z-gap no z-gap support makes something else behave completely differently
It makes a very strong decision on behalf of the user, w/o any identifiable benefits.
It is completely invisible, because it does not give the control to the user.
It does not inform the user, what it does, why it does it.

supermerill, 2020-11-18:
Note that this behaviour may be here in slic3r & derivates since a long time.

> It is inconsistent, in the sense that z-gap no z-gap support makes something else behave completely differently

if you extrude a overhang with overhang flow over a soluble support, you only get passive overextrusion, and if the support are activated, you should have a support below (almost) every overhang.
The problem, imo, is the "support only on build plate" option that doesn't reciprocate into this switch. Not that simple to do.

> It makes a very strong decision on behalf of the user, w/o any identifiable benefits.

* soluble support with bridge flow on overhang => massive overextrusion, likely fail of the print.
* soluble support without bridge flow => expected result as the overhangs are printed flat on the soluble support with normal flow.

> It is completely invisible, because it does not give the control to the user.

Yeah, I'm going to add a message to let the user choose which one he want to disable, between soluble and overhangs.

> It does not inform the user, what it does, why it does it.

same as previously, i don't see the difference between the two criticisms.

kursatu, 2020-11-18:
Here is my preferred behavior, it addresses the issue you wrote. 
1- Use regular flow for overhangs and bridge (Ie. don't over extrude by using the nozzle diameter as the flow height)
2- Always generate overhangs (ie. even for soluble support)
My changes are for these are here, they are trivial.
https://github.com/supermerill/SuperSlicer/compare/master...kursatu:kursatu/Change_overhang_and_bridge_flow?expand=1

On a related note, the people in  #493 discussion found the regular overhang flow to be much better,
Is there any particular reason that it is not changed? 
https://github.com/supermerill/SuperSlicer/issues/493#issuecomment-705247843
https://github.com/supermerill/SuperSlicer/issues/493#issuecomment-703375174

supermerill, 2020-11-18:
ok, I will modify the switch to only remove the flow, not the speed/fan (but imo, they are not needed over soluble supports).

> On a related note, the people in #493 discussion found the regular overhang flow to be much better,
Is there any particular reason that it is not changed?

you can set the overhang flow detection to 0, and it's deactivated, so you have your wanted behavior. Is there something wrong with that?

kursatu, 2020-11-20:
> ok, I will modify the switch to only remove the flow, not the speed/fan (but imo, they are not needed over soluble supports).
I agree mostly, if there is support (soluble or not) then there is no need for speed/fan, but no flow for soluble.
If it is just overhang w/o support then speed/fan/flow settings should take effect. 
Here is with my changes, overhangs are printed with overhang speed/fan.
[image]
Here is the alpha, no speed/fan for overhangs whatso ever.
[image]

> you can set the overhang flow detection to 0, and it's deactivated, so you have your wanted behavior. Is there something wrong with that?
I did not know that, thanks. I'll take that to the other thread , I might have encountered another issue.

supermerill, 2022-03-23:
evrything should be possible now.
reopen if there is still a problem.

closed 2022-04-01 by supermerill