supermerill/SuperSlicer #682: [QUESTION] Line Overlap

qwewer0, 2020-11-18:
### Description
How is the line overlap calculated in PS/SS?
I would like to know the function that the slicer uses to calculate it.

### Example

Extrusion width: 0.4
Body width (left to right): 4mm, 8mm, 12mm ({10, 20, 30} * 0.4)
If the lines would be 0.4 wide without any overlap, then there wouldn't be any gap fill and extra walls/lines, right?
[image]

supermerill, 2020-11-18:
There is a text in the perimeter tab to explain the width of the overall walls.
for more details on why: https://manual.slic3r.org/advanced/flow-math
tl/dr : extrusions are not square in real life.

qwewer0, 2020-11-18:
@supermerill Thank you so much!
With that link I was able to make it to work.

Here are my experiences, if anyone would need it in the future:
(Parameter names has no exact connection to the slicer...)
```
Part_Width = Inside_Line_Width * ( Line_Count - 2 ) + Outside_Line_Width * 2 (If there are two outside line)
Outside_Line_Width = Extrusion_Width - ( ( Layer_Height * ( 1 - ( PI / 4 ) ) ) / 2 )
Inside_Line_Width = Extrusion_Width - ( Layer_Height * ( 1 - ( PI / 4 ) ) )
```
Extrusion_Width = 0.4
Layer_Height = 0.2
Line_Count: Left = 10, Right = 20
Part_Width: Left = 3.6137166941154075 mm, Right = 7.1845130209103036 mm
[image]

closed 2020-11-18 by qwewer0