supermerill/SuperSlicer #683: [Question] What settings should be used to mimic PS Detect Bridging = Off

neophyl, 2020-11-18:
### Version
2.2.55.1 but also in other recent versions

In Prusa Slicer there is a tickbox for Detect Bridging perimeters which when unticked basically just turns off all bridging detection so all perimeters just get printed with normal flows and speeds.  SuSi is missing that option as far as I can see as it has the % options for fan/speeds etc.  However no matter what values I've tried (up to 500%) it never seems to actually completely turn off bridge detection.
What settings should be used to duplicate the detect bridging off of standard PS ?

supermerill, 2020-11-18:
0

neophyl, 2020-11-18:
Thanks.  It IS in the tooltip.  For some reason I completely overlooked that.  I also never even thought about entering 0 on its own as my brain had locked on to it being a % value.

closed 2020-11-18 by neophyl