supermerill/SuperSlicer #684: compilation failing

Tinchus2009, 2020-11-18:
### Version
latest stable release

### Operating system type + version
win10 64

Im trying to compile superslicer, I have tried several times follwing the instructions. Deps build without problems, the rest of the processes finishes ithout errors. But the last final compilation fails with these errors:

"C:\SuperSlicer-master\build\INSTALL.vcxproj" (default target) (1) ->
       (PostBuildEvent target) ->
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.ta
       rgets(153,5): error MSB3073: The command "setlocal [C:\SuperSlicer-master\build\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: "C:\Program Files\CMake\bin\cmake.exe" -DBUILD_TYPE=Release -P cmake_install.cmake [C
       :\SuperSlicer-master\build\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: if %errorlevel% neq 0 goto :cmEnd [C:\SuperSlicer-master\build\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: :cmEnd [C:\SuperSlicer-master\build\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone [C:\SuperSlicer-master\buil
       d\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: :cmErrorLevel [C:\SuperSlicer-master\build\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: exit /b %1 [C:\SuperSlicer-master\build\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: :cmDone [C:\SuperSlicer-master\build\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: if %errorlevel% neq 0 goto :VCEnd [C:\SuperSlicer-master\build\INSTALL.vcxproj]
       C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targ
       ets(153,5): error MSB3073: :VCEnd" exited with code 1. [C:\SuperSlicer-master\build\INSTALL.vcxproj]

    903 Warning(s)
    1 Error(s)

Suggestions?

supermerill, 2020-11-18:
it is with visual studio or msbuild?

Tinchus2009, 2020-11-18:
with msbuild

supermerill, 2020-11-22:
I don't know what's wrong.