supermerill/SuperSlicer #685: How to set up first layer on raft?

ZhenyaKa, 2020-11-18:
### Version
2.3.55

### Operating system type + version
Win64

Slicing on table use first layer height option.
When use raft I got 0.4 layer height of object on raft and can ajust that height.

supermerill, 2020-11-18:
fixed in 2.3.55.1

brianredbeard, 2020-12-15:
@ZhenyaKa Out of curiosity, what is your nozzle diameter?

ZhenyaKa, 2020-12-15:
> @ZhenyaKa Out of curiosity, what is your nozzle diameter?

0,4 and 0.6 I use

liftbag, 2021-01-01:
I'm using the *2.3.55.3 fix error in Voron.ini update ender3 manual mmu C/C++ CI macos #192* (because I cannot use the newer versions. See #777) and seem to me that the issue isn't fixed yet.
From what I see, the raft-piece gap is exactly equal to the diameter of the nozzle + the offset set.
It would be, in my opinion, much more correct and easier to understand and set, that the raft-piece gap followed exactly the same logic used for the support-piece gap, namely layer height + offset.

Limb, 2021-01-26:
Running 2.3.55.4 (commit a4699e7) and also seeing this issue. When setting any number of raft layers, the first layer of the model seems to be set to the nozzle diameter. With a 0.4 nozzle and 2 rafter layers, the first object layer is 0.4 as such:

[image]

All subsequent layers after this one are the correct 0.2mm height. For a 0.6 nozzle, first layer is 0.6:

[image]

Trying to adjust the layer height with adapter layers also shows it as 0.4 and blocked out in red as invariable.

[image]

supermerill, 2021-04-27:
of course, as there is no support to push through, the filament will be a cylinder and so the height will be the diameter of the nozzle (in theory). It may be a bit smaller with lower bridge flow, but I haven't tested it.

> It would be, in my opinion, much more correct and easier to understand and set, that the raft-piece gap followed exactly the same logic used for the support-piece gap, namely layer height + offset.

I agree.
Seems like prusa is rewriting the support code right now, so I'll wait for their 2.4 to touch it again.

closed 2022-04-01 by supermerill