supermerill/SuperSlicer #686: How to setup support max distance from object?

ZhenyaKa, 2020-11-18:
I can't find that option

And I got overlaping raft/support too
http://ipic.su/img/img7/fs/kiss_188kb.1605701266.jpg

supermerill, 2020-11-18:
I don't think it exists

ZhenyaKa, 2020-11-18:
How to solve overlaping raft problem?

supermerill, 2020-11-18:
You should post your project file, i can't do anything with a picture only.

ZhenyaKa, 2020-11-18:
> You should post your project file, i can't do anything with a picture only.

Here
[Knopka.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5560012/Knopka.3mf.zip)

supermerill, 2020-11-18:
I think it's a bug, that shouldn't happen.

ZhenyaKa, 2020-11-18:
> I think it's a bug, that shouldn't happen.
> 
> if you need to print it right now, you can rotate 90° use the cut tool to cut between the two pieces, rotate them again and put them apart a bit more.

I can move them far from each other, but some time need to print small objects near to each other with raft (because them small) and as you can see raft overlaping in that case

neophyl, 2020-11-18:
Its always been that way even with PS and earlier versions.  If you combine the objects into one then it doesn't do that but that is a hack and can cause other issues (like infill patterns not being the same as its now one object for example).  It would be better if support rafts didn't do that.  Hope its not a ton of work.

wolfallein, 2022-01-09:
I don't know if it this is the same problem, but I cannot find where I can change the raft layer height, and the distance between raft and the object. It seems that the raft height is 80% of default layer height and the distance from raft to the object is 100% layer height.

[image]

If I change the **Z-offset Top and/or bottom** it just adds to the default layer height...

If these options are in another place, it would be good to know.

Thanks in advance.

supermerill, 2022-01-26:
the raft height may be the first layer height or the max layer height from the extruder.