supermerill/SuperSlicer #687: Mac OS 2.2.53 error on generate calibration model

vvhh2002, 2020-11-18:
### Version
2.2.53

### Operating system type + version
Macosx 10.15.7

### 3D printer brand / version + firmware version (if known)
VORON 2.4

### Behavior
* Use Calibration
* generate any test model
* 
[image]

supermerill, 2020-11-18:
There is a problem with your start gcode.
can you post it here?

vvhh2002, 2020-11-19:
I use default start G-CODE:

in printer:
```
print_start EXTRUDER={first_layer_temperature[initial_extruder] + extruder_temperature_offset[initial_extruder]} BED=[first_layer_bed_temperature]
```
```
print_end    ;end script from macro
```

```
;BEFORE_LAYER_CHANGE
;[layer_z]
```

```
;AFTER_LAYER_CHANGE
;[layer_z]
```

filament none

supermerill, 2020-11-22:
that print start don't work for 2.2.53, only next versions
you have to use
`print_start EXTRUDER=[first_layer_temperature] BED=[first_layer_bed_temperature]`

closed 2021-06-14 by supermerill