supermerill/SuperSlicer #689: Stuttering and halting while printing

Whitehawk2000, 2020-11-18:
### Version
2.3.55.0 / 2.3.55.1

### Operating system type + version
Win10 latest updates

### 3D printer brand / version + firmware version (if known)
DIY Prusa-Style on 8bit Hardware with Klipper as firmware

### Behavior
When printing, my printer is often stuttering and halting for half a second and more, producing ugly blobs. 
A gcode generated with PrusaSlicer 2.3.0alpha3 and "identical" settings works flawlessly!
I know this behaviour from Marlin-Firmwares on 8-bit hardware when there is not enough processing power.
But due to having Klipper and that it is working with PrusaSlicer gcode, something has to be messed up in the SuperSlicer.

supermerill, 2020-11-18:
do you stream the gcode to the serial port? like with octoprint?

If the problem is too many small gocde, set printer->minimum extrusion length to something like 0.05mm, or greater if it's still too small

Whitehawk2000, 2020-11-18:
Tried even with minimum extrusion length and as expected no change. With Klipper this definitely shouldn't be an issue. And PrusaSlicer works flawless, if there is 0.035 the default value. 

But what i noticed is, that it makes these stops at overhangs and waits until the cooling fans spin up.
So i turned everything cooling related off and at last the setting "overhangs_width_speed" was the cause for the stuttering! So something with this parameter is definetly wrong.

Now i will turn on necessary cooling options like bridging to see if these also have a bad effect.

Whitehawk2000, 2020-11-22:
Finally tested the other cooling settings. They work fine.
Only the "overhangs_width_speed"-parameter makes everything unprintable and is obviously messed up!

Whitehawk2000, 2020-12-08:
With version 2.3.55.2 this problem still exists. Especially on rounded overhangs like on the bottom of the Voron_Design_Cube_v7.stl . With Klipper there is enough computing power to handle high resolution stl's. But even if "minimum extrusion length" is raised by a lot, the problem exists in the same way.
I sliced the same model with the same settings in SuperSlicer and PrusaSlicer.
The gcode generated by PrusaSlicer prints like a charm and SuperSlicer still manages it to somehow mess up the gcode, so that my printer starts to stutter when "overhangs_width_speed" is turned on.
In the next few weeks i am going to test the file on some prusa mk3s (8bit Marlin), Voron0 (32bit), Voron2.4 (32bit), BLV mgn cube (duet2), Sparkcube (32bit RADDS) and maybe a Craftbot flow idex. And i am very sure, that they will suffer of the same problem by the gcode SuperSlicer generates with"overhangs_width_speed" turned on.

supermerill, 2020-12-08:
I just fixed a bug where the settings wasn't used on perimeters.

Whitehawk2000, 2020-12-09:
I have tried the bugfixed version and it seems to work good so far. :-)

Whitehawk2000, 2020-12-15:
Hey, sadly i have to revise my last statement. I recognized, that i made a typo in "overhangs_width_speed" and put it to 55% what made it work with the Voron_Design_Cube_v7.stl
As soon as i put the parameter back to standard 50% the bug returns independently of the stl i am slicing!

So still, the only option is to deactivate the function (parameter 0) until the bug is fixed!

FillPavS, 2021-01-23:
I have the same problem. Klipper on SKR 1.3 and raspberry 3b +. On overhangs that could be printed without special conditions, small areas are generated to which the bridge flow and fan speed are applied. Because of this, the buffer overflows and the head stops for a short period of time, squeezing out the drop. Do I have to provide so you can reproduce the problem? Model, profiles, saved project? Sorry for bad English :)
`
;LAYER:38
; custom gcode end: before_layer_gcode

G1 Z4.750000 F6000.000
G1 E-5.00000 F1800.00000
G1 X104.508 Y29.794 F6000.000
G1 E5.00000 F1800.00000
;TYPE:Support material interface
;HEIGHT:0.2875
G1 F3600.000
G1 X104.508 Y27.548 E0.09618
G1 X105.498 Y27.548 E0.04237
G1 X106.058 Y27.999 E0.03080
G1 X106.542 Y28.448 E0.02828
G1 X106.760 Y28.683 E0.01373
G1 X106.760 Y29.794 E0.04758
G1 X104.568 Y29.794 E0.09388
G1 E-5.00000 F1800.00000
G1 X100.000 Y27.040 F6000.000
G1 E5.00000 F1800.00000
G1 F3600.000
G1 X99.030 Y27.040 E0.04154
G1 X99.011 Y26.683 E0.01531
G1 X99.403 Y26.641 E0.01688
G1 X100.000 Y26.619 E0.02560
G1 X100.597 Y26.641 E0.02560
G1 X100.989 Y26.683 E0.01688
G1 X100.970 Y27.040 E0.01531
G1 X100.060 Y27.040 E0.03897
G1 X100.000 Y27.917 F6000.000
G1 F3600.000
G1 X97.000 Y27.917 E0.12848
G1 X97.000 Y26.247 E0.07151
G1 X97.394 Y26.116 E0.01779
G1 X98.034 Y25.953 E0.02828
G1 X98.684 Y25.836 E0.02828
G1 X99.340 Y25.765 E0.02828
G1 X100.000 Y25.742 E0.02828
G1 X100.660 Y25.765 E0.02828
G1 X101.316 Y25.836 E0.02828
G1 X101.966 Y25.953 E0.02828
G1 X102.606 Y26.116 E0.02828
G1 X103.000 Y26.247 E0.01779
G1 X103.000 Y27.917 E0.07151
G1 X100.060 Y27.917 E0.12591
G1 E-5.00000 F1800.00000
G1 X95.492 Y29.794 F6000.000
G1 E5.00000 F1800.00000
G1 F3600.000
G1 X93.240 Y29.794 E0.09644
G1 X93.240 Y28.683 E0.04758
G1 X93.458 Y28.448 E0.01373
G1 X93.942 Y27.999 E0.02828
G1 X94.502 Y27.548 E0.03080
G1 X95.492 Y27.548 E0.04237
G1 X95.492 Y29.734 E0.09362
G1 E-5.00000 F1800.00000
G1 X103.631 Y30.671 F6000.000
G1 E5.00000 F1800.00000
G1 F3600.000
G1 X103.631 Y28.794 E0.08039
G1 X96.369 Y28.794 E0.31103
G1 X96.369 Y30.671 E0.08039
G1 X92.363 Y30.671 E0.17157
G1 X92.363 Y28.338 E0.09994
G1 X92.838 Y27.827 E0.02987
G1 X93.368 Y27.335 E0.03096
G1 X93.931 Y26.882 E0.03096
[... 1608 lines cut ...]
G1 X143.266 Y85.776 E0.01516
G1 X143.858 Y85.885 E0.01888
G1 X144.441 Y86.032 E0.01886
G1 X145.362 Y86.345 E0.03049
G1 X145.929 Y86.591 E0.01937
G1 X146.759 Y87.031 E0.02946
G1 X147.294 Y87.373 E0.01992
G1 X147.898 Y87.823 E0.02361
G1 X148.467 Y88.319 E0.02364
G1 X148.992 Y88.853 E0.02349
G1 X149.479 Y89.431 E0.02371
G1 X149.709 Y89.737 E0.01200
G1 X150.125 Y90.366 E0.02365
G1 X150.491 Y91.024 E0.02359
G1 X150.805 Y91.708 E0.02359
G1 X151.066 Y92.414 E0.02359
G1 X151.272 Y93.141 E0.02372
G1 X174.790 Y190.025 E3.12594
G1 X174.790 Y197.290 E0.22779
G1 X100.060 Y197.290 E2.34311
G1 X99.800 Y196.944 F6000.000`

[image]
[image]
[image]

FillPavS, 2021-01-23:
> I just fixed a bug where the settings wasn't used on perimeters.

2.3.55.2
I have the same problem. Klipper on SKR 1.3 and raspberry 3b +. On overhangs that could be printed without special conditions, small areas are generated to which the bridge flow and fan speed are applied. Because of this, the buffer overflows and the head stops for a short period of time, squeezing out the drop. Do I have to provide so you can reproduce the problem? Model, profiles, saved project? Sorry for bad English :)

<details>
  <summary>GCODE bug Layer</summary>
;LAYER:38
; custom gcode end: before_layer_gcode

G1 Z4.750000 F6000.000
G1 E-5.00000 F1800.00000
G1 X104.508 Y29.794 F6000.000
G1 E5.00000 F1800.00000
;TYPE:Support material interface
;HEIGHT:0.2875
G1 F3600.000
G1 X104.508 Y27.548 E0.09618
G1 X105.498 Y27.548 E0.04237
G1 X106.058 Y27.999 E0.03080
G1 X106.542 Y28.448 E0.02828
G1 X106.760 Y28.683 E0.01373
G1 X106.760 Y29.794 E0.04758
G1 X104.568 Y29.794 E0.09388
G1 E-5.00000 F1800.00000
G1 X100.000 Y27.040 F6000.000
G1 E5.00000 F1800.00000
G1 F3600.000
G1 X99.030 Y27.040 E0.04154
G1 X99.011 Y26.683 E0.01531
G1 X99.403 Y26.641 E0.01688
G1 X100.000 Y26.619 E0.02560
G1 X100.597 Y26.641 E0.02560
G1 X100.989 Y26.683 E0.01688
G1 X100.970 Y27.040 E0.01531
G1 X100.060 Y27.040 E0.03897
G1 X100.000 Y27.917 F6000.000
G1 F3600.000
G1 X97.000 Y27.917 E0.12848
G1 X97.000 Y26.247 E0.07151
G1 X97.394 Y26.116 E0.01779
G1 X98.034 Y25.953 E0.02828
G1 X98.684 Y25.836 E0.02828
G1 X99.340 Y25.765 E0.02828
G1 X100.000 Y25.742 E0.02828
G1 X100.660 Y25.765 E0.02828
G1 X101.316 Y25.836 E0.02828
G1 X101.966 Y25.953 E0.02828
G1 X102.606 Y26.116 E0.02828
G1 X103.000 Y26.247 E0.01779
G1 X103.000 Y27.917 E0.07151
G1 X100.060 Y27.917 E0.12591
G1 E-5.00000 F1800.00000
G1 X95.492 Y29.794 F6000.000
G1 E5.00000 F1800.00000
G1 F3600.000
G1 X93.240 Y29.794 E0.09644
G1 X93.240 Y28.683 E0.04758
G1 X93.458 Y28.448 E0.01373
G1 X93.942 Y27.999 E0.02828
G1 X94.502 Y27.548 E0.03080
G1 X95.492 Y27.548 E0.04237
G1 X95.492 Y29.734 E0.09362
G1 E-5.00000 F1800.00000
G1 X103.631 Y30.671 F6000.000
G1 E5.00000 F1800.00000
G1 F3600.000
G1 X103.631 Y28.794 E0.08039
G1 X96.369 Y28.794 E0.31103
G1 X96.369 Y30.671 E0.08039
[... 1614 lines cut ...]
G1 X143.858 Y85.885 E0.01888
G1 X144.441 Y86.032 E0.01886
G1 X145.362 Y86.345 E0.03049
G1 X145.929 Y86.591 E0.01937
G1 X146.759 Y87.031 E0.02946
G1 X147.294 Y87.373 E0.01992
G1 X147.898 Y87.823 E0.02361
G1 X148.467 Y88.319 E0.02364
G1 X148.992 Y88.853 E0.02349
G1 X149.479 Y89.431 E0.02371
G1 X149.709 Y89.737 E0.01200
G1 X150.125 Y90.366 E0.02365
G1 X150.491 Y91.024 E0.02359
G1 X150.805 Y91.708 E0.02359
G1 X151.066 Y92.414 E0.02359
G1 X151.272 Y93.141 E0.02372
G1 X174.790 Y190.025 E3.12594
G1 X174.790 Y197.290 E0.22779
G1 X100.060 Y197.290 E2.34311
G1 X99.800 Y196.944 F6000.000
</details>

[image]
[image]
[image]

supermerill, 2021-03-19:
@FillPavS you can change the overhang threshold just a little bit to avoid that problem. But in a circle, you'll always have tiny segments.

DasBootU96, 2021-05-21:
@FillPavS FWIW : Using the DisplayLayerProgress Plugin ? If yes, it causes hostbuffer saturation. uninstall DisplayLayerProgress , slice new gcode and print again. Check https://github.com/supermerill/SuperSlicer/issues/369

closed 2021-06-04 by supermerill