supermerill/SuperSlicer #690: Toolchange temperature

Tinchus2009, 2020-11-18:
### Version
2.2.53 stable

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
custom corexy reprapfirmware dual extruder printhead

### Behavior
Im setting up a dual extrusion print and trying to setup the PVA parameters. The Toolchange temperature value, Im trying to set it up to 120 degrees, but it always defaults me to 175, wich is too high (pva ca even be printed at that temp)

supermerill, 2020-11-18:
"Toolchange temperature" is the temp for extracting the current filament from the nozzle.

Tinchus2009, 2020-11-19:
So where do you setup the standby temperature for a dual extrusion?

supermerill, 2020-11-19:
mmm....
can you post your project file, and the lines that are wrong in the gcode?

bakaufman, 2020-12-11:
I have questions about setup of dual extruder as well.