supermerill/SuperSlicer #691: Creating a Bridge Flow calibration with 15% steps and 6 tests causes SuperSlicer to crash

donicrosby, 2020-11-19:
### Version
2.2.53

### Operating system type + version
Windows 10 (1904)

### 3D printer brand / version + firmware version (if known)
Prusa MK3S (Firmware 3.9.2)

### Behavior
When you try to create a bridge flow calibration object test for with 15% steps and 6 tests causes SuperSlicer to crash (it can't find the f5.amf bridge flow calibration project)

[image]
[image]

#### Project File (.3MF) where problem occurs
N/A built in calibration object (see above)

supermerill, 2020-11-19:
thinks for the bug report, I have to add some failsafe.

Testing with 5% flow is a bit extreme.

supermerill, 2020-11-21:
I can't reproduce, can you send me your project file?

supermerill, 2020-11-21:
sorry, that was resolved in the alpha version, but I forgot to push it back on the stable.

donicrosby, 2020-11-24:
Do you want me to close this? Or should I wait until 2.3.x is released to verify before closing?

supermerill, 2020-11-25:
you can verify it works and close it, I'lll close it when doing the next cleaning pass

closed 2021-04-27 by supermerill