supermerill/SuperSlicer #695: Seam pain with >1 model on plater leads to crash

mocksoul, 2020-11-20:
### Version
2.3.55.1 ("slimfast")

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
voron 2.4/klipper

### Behavior
Add >1 model to print, try to use "seam paint" on one, slice => program crash.

Seam painting for 1 model being slicing seems to work fine.

#### Project File (.3MF) where problem occurs
[silver_abs_seam_paint_crash.zip](https://github.com/supermerill/SuperSlicer/files/5572111/silver_abs_seam_paint_crash.zip)

(painted seam on fan cover (bottom left). Just open and try to slice - crash)

supermerill, 2020-11-21:
good thing: now (next nightly) the .3mf can be opened with the prusaslicer
 * superslicer settings are now saved in a other file inside it
 * the prusaslicer setting file is filled with converted settings (best effort)

bad thing: the bug is also present on prusaslicer, so it should be reported on their issue tracker, so they can fixed it for their beta release.

[silver_abs_seam_paint_crash_2.zip](https://github.com/supermerill/SuperSlicer/files/5578229/silver_abs_seam_paint_crash_2.zip)

supermerill, 2021-04-27:
reopen if it wasn't fixed in the next release / current nightly.

closed 2021-04-27 by supermerill