supermerill/SuperSlicer #696: Acceleration / jerk for each move type

espr14, 2020-11-20:
### Version
v2.2.54

### Operating system type + version
all

### 3D printer brand / version + firmware version (if known)
all

### Behavior
User can set different speed for different types of movement (external/internal perimeter, infill, support etc.). But it's not just the speed that influences the quality of the print. It's also acceleration and (Marlin) jerk (min. speed, not the physical jerk - mm/s^3). But there's no reason to decrease global acceleration due to external perimeter because it will cause stringing during traveling.

Can you add acceleration and jerk for each type of movement? Since there are acc. and jerk for each axis, there could be just a coefficient (in %) that will decrease both. Would be nice to add junction deviation as well.

I've read this is already available in Cura.

#### Project File (.3MF) where problem occurs

supermerill, 2020-11-21:
* jerk : duplicate of #237
for the acceleration, what is missing from here? 
[image]

espr14, 2020-11-21:
1. Internal vs external perimeter acc.
2. Support acc.
3. Travel acc.
4. Maybe thin extrusion acc.
5. First layer acc.
6. Small perimeter acc.

Would be great to have types in rows and speed/acc./jerk as columns.

supermerill, 2020-11-21:
Adding some new accelerations seems rather easy.

espr14, 2020-11-22:
Printing acceleration defines corner adhesion. But ringing is caused by motors so you want to limit motor acceleration as well.

supermerill, 2020-11-22:
> Printing acceleration
> motor acceleration

what's the difference? there is a gcode for each?

espr14, 2020-11-22:
M201 X1000 Y1000
vs
M204 P800

Motor acceleration is applied to motors directly while printing acceleration is applied to change in direction and speed of printing. Motors with 1000 acc. will achieve 1414 acc. in the diagonal direction.

supermerill, 2020-11-22:
So M204 P800 in the machine limits,
and M201 X1000 Y1000 at each feature change?

and travel acceleration is separate, right?

espr14, 2020-11-22:
All at each feature change. It will create a big matrix and lot of these will be defaults. But you cannot avoid it unless you define tags/modifiers and their hierarchy. E.g.

external_perimeter: acceleration: travel: 3000
external_perimeter: acceleration: printing: 800
external_perimeter: speed: printing: 50

Then you can define sparse matrix more efficiently.

x0rtrunks, 2020-11-27:
M204 would be the preferred way to change acceleration by the user. M201 is the firmware limit.

espr14, 2020-11-27:
Jerk and M201 have impact on ringing in the first place. The other ways are just hotfixes. This is where FW sets limits (Marlin):
```
#define LIMITED_MAX_ACCEL_EDITING     // Limit edit via M201 or LCD to DEFAULT_MAX_ACCELERATION * 2
#if ENABLED(LIMITED_MAX_ACCEL_EDITING)
  #define MAX_ACCEL_EDIT_VALUES       { 6000, 6000, 200, 20000 } // ...or, set your own edit limits
#endif
```

astorun, 2020-12-20:
I would really love to see this feature. Also seperate top solid layer accel/jerk will be awesome!

richfelker, 2021-01-17:
External perimeter acceleration is extremely important and is probably the blocking issue for me switching from Cura to SuperSlicer. Limiting all perimeters to low acceleration makes prints take forever, but high acceleration on external perimeter causes cosmetic and dimensional accuracy problems due to ringing/backlash.

richfelker, 2021-01-17:
Limiting individual motor acceleration should not be done in the gcode emitted by the slicer, or should at least be optional. It's nontrivial how such limits even interact with anything but a straight cartesian printer (e.g. delta, corexy, etc.) and in my experience everyone just treats them as static hard limits for the machine, not as settings to tune dynamically. I wouldn't even be surprised if Marlin has bugs (esp when using JD) with them changing dynamically.

VanessaE, 2021-01-20:
I would like to put my vote in for travel acceleration.

Or if not that, then at the very least, the current behavior should remain, i.e. that no `M204 Txxxx` commands are ever put into the g-code.  At least then the user can still just set it once and then let it be saved in EEPROM.

richfelker, 2021-01-20:
Yes, travel acceleration is important too. I only overlooked it because I use the same (max the machine can reliably handle) acceleration for travel and all parts of print except outer-perimeters, so `default_acceleration` suffices to get the right result for it.

richfelker, 2021-01-20:
For what it's worth, just setting `M204` with `P` and `T` rather than `S` should suffice to get high travel acceleration, but at least Klipper firmware does not honor separate settings here, so having the slicer emit separate `M204 S...` for each change between types of moves is more reliable.

zhechenzhu, 2021-06-10:
+1 for more control over acceleration of different move types. Lower external perimeter acceleration for quality and higher internal for speed.

AndrewEllis93, 2021-10-04:
+1 to this. We are currently working around this with some post-processing scripts and extrusion role change custom gcode, which works but is not ideal.

julianschill, 2021-10-30:
I am currently trying to implement this. This is how it looks like.
[image]
My branch can be found here: https://github.com/julianschill/SuperSlicer/tree/julianschill_better_acceleration_control

Is this enough or is there a need to be able to configure accelerations for every type? Is something useful missing? 
 
I am also considering refactoring the whole accelerations code  to a switch/case statement to make it more explicit and better understandable: https://github.com/supermerill/SuperSlicer/blob/403d258c1c5d8cc06a1e608ad7b2c0f661829ef0/src/libslic3r/GCode.cpp#L3867-L3883

Once it's done I could create a pull request, if there is interest.

zhechenzhu, 2021-10-31:
Excellent work. Missing these coming from Cura. Hope to see them merged soon.

closed 2022-04-01 by supermerill