supermerill/SuperSlicer #698: Proper autoarrange?

rkarlsba, 2020-11-20:
### Version
2.2.54

### Operating system type + version
Irrelevant

### 3D printer brand / version + firmware version (if known)
Irrelevant

### Behavior
Autoarrange ignores shape of object and just autoarranges objects as if they were enclosed by rectangles

#### Project File (.3MF) where problem occurs
http://karlsbakk.net/tmp/sikksakk.3mf

PS: An attempt to upload a 3mf file fails with 'We don't support that file type. Try again with a GIF, JPEG, JPG, PNG, DOCX, GZ, LOG, PDF, PPTX, TXT, XLSX or ZIP."

neophyl, 2020-11-20:
Github does not allow 3mf files which is why they are normally zipped up.  That's normal.  Just zip it up like the rest of us.

wildjokerss, 2020-11-20:
Believe this is a duplicate of https://github.com/supermerill/SuperSlicer/issues/526

rkarlsba, 2020-11-20:
@neophyl, sure, 3mf is a zip file, after all, so why not zip it again? or why not fix the problem in the first place?
@wildjokerss, it certainly is a duplicate of that one, yes.

neophyl, 2020-11-20:
rkarlsba, I only point it out as this is github and I think what file types are allowed is determined by them and not by Supermerill.  As such asking him to 'fix' the issue is a bit out of whats possible.  You could always message the site maintainers and ask them to add it to the list of allowed file types.  

Or just change the file type ending. But that tends to confuse people unless you point that out EVERY time you post a 3mf.  Its just easier to to zip it up (yes again) as people know to unzip a zip file.