supermerill/SuperSlicer #699: Slicer doesn't remeber last folder for opened file and for expored gcode.

ZhenyaKa, 2020-11-20:
Slicer doesn't remeber last used folder for opened files) and for expored gcode.

Need each time to select folder :(

### Version
2.3.55

### Operating system type + version
Win 7 X64

neophyl, 2020-11-20:
I can only report on what v 2.2.55.1 is doing for me on Windows 10.
If I run the software, with a new blank plater and select add part, it opens the last used folder I imported a part from.  If I then navigate to a different folder and load in a stl then exit the program and restart and repeat the add part it opens with the last selected one.  So that is different to your experience.  Same with opening a 3mf project, it opens the last folder I used.

What sort of permissions does the software running under ?  If It cant save to the configuration folder it cant save the last used settings.  Have you tried running it as Administrator ?  Do your settings folders have any sort of restrictions on them ?

Lastly as a test if you extract the software to a folder on a USB drive and then run it from the superslicer_local-settings.bat shortcut in the extracted folder, it should start as if its a clean install, ie no profiles, go through the wizard process etc.  Once you do that and load a part in does it remember then ?  The theory is a USB drive shouldnt have any restrictions on it so it shouldnt have an issue remembering the last used folder.

ZhenyaKa, 2020-11-20:
I press Import (stl etc) end select some folder folder (I was in c:\one, select c:\one\two)
I press Import again and now I not in that selected folder but in folder there I was before selection (c:\one)

TD5023, 2020-11-20:
I've found that this happens when I import/export a config file. Every subsequent import will default to the location where the config is saved. However, if I delete the directory, then it behaves like normal again. I work around this by creating a dummy folder on my desktop, exporting a config bundle to it, and then deleting the folder.

supermerill, 2020-11-21:
Prusa has added a bool "use imperial unit", and this cause some sort of problems, as i also added an other bool for "update last dir", so there is a line where the arguments where wrong.

You can blame it on Usa.

closed 2021-04-27 by supermerill