supermerill/SuperSlicer #701: Add support for MoonsailOS the same way that octoprint is integrated.

vidpugelj, 2020-11-22:
### Version
2.2.53

### Operating system type + version
Irrelevenat

### 3D printer brand / version + firmware version (if known)
Twotrees Sapphire Plus, Klipper+MonsailOS

Any way you integrate monsailos, so you can send files directly to it, the same as with Octoprint?

supermerill, 2020-11-22:
I need to know how to send it to it.

here is the octoprint code
https://github.com/supermerill/SuperSlicer/blob/master/src/slic3r/Utils/OctoPrint.hpp
https://github.com/supermerill/SuperSlicer/blob/master/src/slic3r/Utils/OctoPrint.cpp

here is the duet code
https://github.com/supermerill/SuperSlicer/blob/master/src/slic3r/Utils/Duet.hpp
https://github.com/supermerill/SuperSlicer/blob/master/src/slic3r/Utils/Duet.cpp

you can also look at the repetier one.

Ideally, if you can write your own, do it and submit the changes.

pheonix3330, 2020-11-22:
MainsailOS currently supports the Octoprint implementation for uploading files from a slicer. Just set up Superslicer like you are using Octoprint, put in the IP address of Mainsail and put anything in the password field (it doesn't matter what, I type in "1"). It should work just like Octoprint at that point.

supermerill, 2020-11-22:
I prusa update the octoprint implementation and i didn't notice it, tell me and i'll bring back the old one for MainsailOS.

pheonix3330, 2020-11-22:
> I prusa update the octoprint implementation and i didn't notice it, tell me and i'll bring back the old one for MainsailOS.

It still works in the latest version of Prusaslicer also, I just checked (PrusaSlicer 2.3.0-alpha4). No need to roll anything back.

vidpugelj, 2020-11-22:
Already supported.

closed 2020-11-22 by vidpugelj