supermerill/SuperSlicer #702: Post Processing Directory

ETE-Design, 2020-11-22:
Would be great if you could add an folder under Resources for Post- Processing Script, so you don't have to to specifi full path for each Script...

Something like you only have to tell sub-directory like ./xxx/post-script.bat

supermerill, 2020-11-22:
You can add this directory to your PATH, in your user setting, so the command can be found.

ETE-Design, 2020-11-22:
@supermerill So where in SuperSlicer do I set that PATH? I know it can be set in Windows, but not very convinient for multi platform support...
Would also be nice to have a place to put the Script's inside SuperSlicer...

supermerill, 2020-11-22:
Yes, on windows it's in your user dir.
in linux, on your .bashrc
etc...

>Would also be nice to have a place to put the Script's inside SuperSlicer...

It execute a command. I can't do that. If your command is "python ~/slicer/script/myscript" I can't add "~/slicer/script/" before "python"

ETE-Design, 2020-11-22:
@supermerill Okey, if it isen't possible... Then we can just close it :-)

closed 2020-11-22 by ETE-Design